vcc-gnd/YD-ESP32-S3 #4: YD-EPS32-23 2022-V1.3 Development Board RGB LED Not working

amin-is, 2023-07-04:
Can you give an example code for this board RGB LED blinking OR NeoPixel example code

vcc-gnd, 2023-07-05:
You should have purchased pirated hardware, but I can provide you with technical support.

W2812 Micropython CODE：
from machine import Pin
from neopixel import NeoPixel
#ESP32-S3 48PIN ESP32-C3 8PIN ESP32 16PIN ESP32-S2 18PIN
pin = Pin(48, Pin.OUT)
np = NeoPixel(pin, 1)
np[0] = (10,0,0) #R B G max 255 255 255
np.write()
r, g, b = np[0]

amin-is, 2023-07-05:
Can you give me arduino c language example. I have purchase a board when backside printed was only Board model, version and your web address and no any others information like [this board ]([image]). May be this board is copy..

buxiua, 2023-07-24:
@amin-is Hey guy，I have the same problem but I have fixed it，your board may be a knockoff，please see #1 , and this is my program to blink RGB using arduino。
```C
#include <Arduino.h>
#include <FastLED.h>

#define LED_PIN 48
#define LED_NUMS 1

CRGB leds[LED_NUMS];

void setup() {
  Serial.begin(115200);
  FastLED.addLeds<NEOPIXEL, LED_PIN>(leds, LED_NUMS);
}

void loop() {
  leds[0] = CRGB(255, 0, 0);
  FastLED.show();
  Serial.print("RED \n");
  delay(1000);
  leds[0] = CRGB(0, 255, 0);
  FastLED.show();
  Serial.print("Green \n");
  delay(1000);
  leds[0] = CRGB(0, 0, 255);
  FastLED.show();
  Serial.print("Blue \n");
  delay(1000);
}
```
of course it work correctly!

vcc-gnd, 2023-09-07:
你购买的板子可能为山寨板子，注意通信线路是否连接

dzalf, 2023-11-16:
For newcomers and those having issues with the onboard NEOPIXEL, this is the code from the response above  (by @vcc-gnd ) using code tags so that you can easily copy-paste in your sketch. Remember to plug in your USB-C cable in the port on the right-hand side.

Thanks, guys from @vcc-gnd for making this fantastic version of the ESP32-S3. It is a powerhouse!

``` cpp
/**
   @file: ESP32-S3_DevModule_RGB_test.ino Board from http://vcc-gnd.com
   @brief: Basic onboard WS2012 - NEOPIXEL test
            Driving pin: 48
   @deps: <FastLED> library: https://github.com/FastLED/FastLED

*/

#include <Arduino.h>
#include <FastLED.h>

#define LED_PIN 48
#define LED_NUMS 1

CRGB leds[LED_NUMS];

void setup() {

  Serial.begin(115200);
  FastLED.addLeds<NEOPIXEL, LED_PIN>(leds, LED_NUMS);
}

void loop() {

  leds[0] = CRGB(255, 0, 0);
  FastLED.show();

  Serial.print("RED \n");

  delay(1000);

  leds[0] = CRGB(0, 255, 0);
  FastLED.show();

  Serial.print("Green \n");

  delay(1000);

  leds[0] = CRGB(0, 0, 255);
  FastLED.show();

  Serial.print("Blue \n");

  delay(1000);

}

```

TechRelief, 2023-12-22:
F.Y.I. This board has a solder jumper marked RGB right next to the RBG LED which you have to short with a blob of solder and then the standard RGB Arduino sample code will work.  The RGB led is connected to GPIO pin 38 and I guess the jumper is there in case you want to use this pin for a different purpose.