vcc-gnd/YD-ESP32-S3 #13: Micropython下内置温度和霍尔传感器无法使用

anders0821, 2024-04-08:
刷了Micropython固件：YD-ESP32-S3-N16R8-MPY-V1.1.bin
内置温度和霍尔传感器无法使用，esp32模块存在，但没有函数hall_sensor()和raw_temperature()

>>> import esp32
>>> esp32.hall_sensor()
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
AttributeError: 'module' object has no attribute 'hall_sensor'
>>> esp32.raw_temperature()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'raw_temperature'

genicsoft, 2024-06-08:
你应该去问Micropython 或者用最新版mpy固件

goldenfreemanchina, 2024-12-11:
`>>>import esp32
print(dir(esp32))
`

输出是这样的：
```
MPY: soft reboot
['__class__', '__name__', 'HEAP_DATA', 'HEAP_EXEC', 'NVS', 'Partition', 'RMT', 'ULP', 'WAKEUP_ALL_LOW', 'WAKEUP_ANY_HIGH', '__dict__', 'gpio_deep_sleep_hold', 'idf_heap_info', 'mcu_temperature', 'wake_on_ext0', 'wake_on_ext1', 'wake_on_touch', 'wake_on_ulp']
```

你要的是esp32.mcu_temperature()；

goldenfreemanchina, 2024-12-11:
> 刷了Micropython固件：YD-ESP32-S3-N16R8-MPY-V1.1.bin 内置温度和霍尔传感器无法使用，esp32模块存在，但没有函数hall_sensor()和raw_temperature()
>
> ```
> Traceback (most recent call last):
>   File "<stdin>", line 2, in <module> [...]

```
>>>import esp32
print(dir(esp32))
```

输出是这样的：
```
MPY: soft reboot
['__class__', '__name__', 'HEAP_DATA', 'HEAP_EXEC', 'NVS', 'Partition', 'RMT', 'ULP', 'WAKEUP_ALL_LOW', 'WAKEUP_ANY_HIGH', '__dict__', 'gpio_deep_sleep_hold', 'idf_heap_info', 'mcu_temperature', 'wake_on_ext0', 'wake_on_ext1', 'wake_on_touch', 'wake_on_ulp']
```

你要的是esp32.mcu_temperature()；